pibooth/pibooth #549: Error oops something ...

Syaanure, 2024-02-05:
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ DEBUG   ] pibooth           : Preview resolution is (640.0, 480.0)
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'zTXt' 41 4039
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 4092 390
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 4494 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 4494 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 4512 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 4533 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 4533 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 4552 8192
[ DEBUG   ] pibooth           : took 1.671 seconds
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : took 2.162 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : took 4.180 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Can not get camera preview image
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 98, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 501, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 119, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 473, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 99, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 61, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 470, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 138, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 102, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 71, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 155, in preview
    self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 101, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ DEBUG   ] pibooth           : took 0.041 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.21-v8+', version='#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023', machine='aarch64', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.036 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'

Syaanure, 2024-02-05:
I have RPI3 B+ and camera with csi connector.

the camera work with libcamera-still.

werdeil, 2024-02-05:
Hi @Syaanure do you confirm that you are running on RaspiOS Buster? Pibooth is not yet compatible with Bullseye+ version (using libcamera) see #540

Syaanure, 2024-02-06:
Hi @werdeil thank you that works.

closed 2024-02-06 by Syaanure