pibooth/pibooth #550: Reset printing queue

Kourounator, 2024-02-05:
Hello,
Is it possible to remove all the pictures that wait for printing. I have 24 pictures waiting due to children playing at the end of a party 😄
Thanks for your help

werdeil, 2024-02-05:
Hello @Kourounator, once the printing is asked the queue is handled by cups so to clear the printing queue you need to look there.

From what I found [here](https://josetips.wordpress.com/2016/07/03/how-to-clear-the-cups-print-queue/), you need to launch command: `lpstat -o` to see the list of elements in the queue then `cancel -a xxxxx` on all the elements you want to cancel.

Hope it helps!

Kourounator, 2024-02-05:
Hi, thanks for your help. I juste succeed by removing all the printing tasks directly in the OS. It IS ont useful to open thé Pibooth app. 😉

closed 2024-02-10 by werdeil

tekilaz, 2025-03-14:
i've wrote a plugin to add a shortcut directly in pibooth. Please check in open issue.