pibooth/pibooth #555: "Oops something went wrong" when I try to take a photo

Louloubiwan, 2024-02-28:
**Hi everyone**
I am on raspberry pi 5,
So when I choose a layout, there is a little freez befors this message apear : "Oops something were wrong"

I try to use l`ibcamera-still` and `libcamera-hello --list-cameras and the camera still working
`

**Logs**
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : took 4.038 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Can not get camera preview image
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/states.py", line 98, in set_state
    hook(cfg=self.cfg, app=[self.app](http://self.app/), win=self.win)
  File "/usr/local/lib/python3.11/dist-packages/pluggy/_hooks.py", line 501, in __call__
    return self._hookexec([self.name](http://self.name/), self._hookimpls.copy(), kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy/_manager.py", line 119, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy/_manager.py", line 473, in traced_hookexec
    return outcome.get_result()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy/_result.py", line 99, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.11/dist-packages/pluggy/_result.py", line 61, in from_call
    result = func()
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy/_manager.py", line 470, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pluggy/_callers.py", line 138, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/usr/local/lib/python3.11/dist-packages/pluggy/_callers.py", line 102, in _multicall
    res = hook_impl.function(*args)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pibooth/plugins/camera_plugin.py", line 71, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/opencv.py", line 155, in preview
    self._window.show_image(self._get_preview_image())
                            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/opencv.py", line 101, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ DEBUG   ] pibooth           : took 0.002 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.0-rpi8-rpi-2712', version='#1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25)', machine='aarch64', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************

**Environment**
 - OS : Raspberry pi imager
 - Camera : https://www.amazon.fr/dp/B09D7SW9M8?starsLeft=1&ref_=cm_sw_r_cso_wa_apan_dp_4DCBFRASRGR4RSR3GV94
 - Pibooth version : 2.0.8

If anyone can help me or give me information, I will take it with thanksfull !

ferranponce, 2024-02-29:
same problem

werdeil, 2024-03-01:
Hello @Louloubiwan and @ferranponce, Pibooth is not compatible with the latest Raspberry PI OS. The issue you are getting comes from the fact that you are using Bullseye or Bookworm.

We don't known when we will compatible but the issue is followed by #540

Louloubiwan, 2024-03-01:
Hello @werdeil  , Thank you for your reply, do you have a stable OS that you recommend for using the pibooth?

werdeil, 2024-03-02:
As listed in [the documentation](https://pibooth.readthedocs.io/en/stable/sources/install.html#software), we recommend using RaspiOS Buster

closed 2024-05-14 by werdeil