pibooth/pibooth #559: Cannot install requirements from documentation

sarns, 2024-03-07:
**Describe the bug**
I set up a RPi4 with a fresh buster image (2021-05-07). I upgraded all software described in step 3 of the documentation. In step 4 I tried to install libsdl2-* and this step fails.

**Logs**

sudo apt-get install libsdl2-*
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libsdl2-mixer-dev' for glob 'libsdl2-*'
Note, selecting 'libsdl2-image-dev' for glob 'libsdl2-*'
Note, selecting 'libsdl2-gfx-dev' for glob 'libsdl2-*'
Note, selecting 'libsdl2-gfx-doc' for glob 'libsdl2-*'
Note, selecting 'libsdl2-mixer-2.0-0' for glob 'libsdl2-*'
Note, selecting 'libsdl2-dbg:armhf' for glob 'libsdl2-*'
Note, selecting 'libsdl2-dev' for glob 'libsdl2-*'
Note, selecting 'libsdl2-doc' for glob 'libsdl2-*'
Note, selecting 'libsdl2-ttf-dev' for glob 'libsdl2-*'
Note, selecting 'libsdl2-net-2.0-0' for glob 'libsdl2-*'
Note, selecting 'libsdl2-net-dev' for glob 'libsdl2-*'
Note, selecting 'libsdl2-image-2.0-0' for glob 'libsdl2-*'
Note, selecting 'libsdl2-2.0-0-dbgsym:armhf' for glob 'libsdl2-*'
Note, selecting 'libsdl2-2.0-0' for glob 'libsdl2-*'
Note, selecting 'libsdl2-gfx-1.0-0' for glob 'libsdl2-*'
Note, selecting 'libsdl2-ttf-2.0-0' for glob 'libsdl2-*'
libsdl2-2.0-0 is already the newest version (2.0.9+dfsg1-1+deb10u1).
libsdl2-2.0-0 set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libsdl2-2.0-0-dbgsym:armhf : Depends: libsdl2-2.0-0:armhf (= 2.0.9+dfsg1-1+rpt1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

**Environment**
 - OS "Debian GNU/Linux 10 (buster)" on RPi 4
 - Logitech C920
 - Pibooth version [e.g. 1.1.2]

Lachemoitumsoule, 2024-03-08:
I have the same probleme who use Buster 64bit. That's works only in Buster 32bit...

sarns, 2024-03-08:
Thanks for the hint. I switched to 32 bit Buster and now it worked fine.

Added the hint to documentation: https://github.com/pibooth/pibooth/pull/560

closed 2024-03-08 by sarns