pibooth/pibooth #561: Soundcard muted

Liedzki, 2024-03-10:
Hello,

I wanted to play some music, when the photo booth is in use. I wanted to use an audio player, like audacious, to play music. 
I added a iquaudio digiamp+ on my raspberry pi 4, but if I start pibooth, the soundcard is muted. If pibooth is closed the, the soundcard keeps muted. It is unmuted if I restart raspian. 

I'm not familiar with programming, but with a little guidance I'm confident I can implement a solution.

Thank you in advance!

werdeil, 2024-06-15:
Indeed, even if no sound is used by pibooth (only by [pibooth-sound-effects](https://github.com/pibooth/pibooth-sound-effects)) the soundcard is initialized by pygame when we run `pygame.init()`

I need to test if releasing the mixer right after in not creating issues on the [pibooth-sound-effects](https://github.com/pibooth/pibooth-sound-effects) plugin.

werdeil, 2024-06-15:
I just pushed something that may fix the issue but I'm not sure as I didn't manage to reproduce it (I guess it happens only with an external soundcard).

This fix will be included in next release.