pibooth/pibooth #566: Hybrid mode issue

TboneSteak237, 2024-04-01:
Hi, i am having issues using hybrid mode. all works well with just the DSLR connected. 
The camera doesn't support live view (i have tried a separate DSLR that does have live preview but its too laggy, which i assume is related to resolution.) which is why i have a pi camera, well a 3rd party camera, connected to try utilize hybrid mode, but never got it to work.

Raspberry pi 4B 4GB
OS: Bullseye 32Bit Legacy
DSLR: Canon 400D
Camera: Arducam OV5647

I can get an image with `libcamera-hello -t 0` so i know all the hardware is functional.

`libcameraa-hello -t 0 --nopreview`:

> [0:10:10.124278262] [1660]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:10:10.169421336] [1661]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media3
[0:10:10.169511800] [1661]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:10:10.170752006] [1660]  INFO Camera camera.cpp:1033 configuring streams: (0) 1296x972-YUV420
[0:10:10.171313237] [1661]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA

`libcamera-still --list-cameras`

> Available cameras

> -----------------
> 0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [30.00 fps - (0, 0)/0x0 crop]
                             1296x972 [30.00 fps - (0, 0)/0x0 crop]
                             1920x1080 [30.00 fps - (0, 0)/0x0 crop]
                             2592x1944 [30.00 fps - (0, 0)/0x0 crop]

I have updated everything with `sudo apt-get update` and `sudo apt-get full-upgrade

Here is the log:

> [ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ DEBUG   ] pibooth           : Try to kill process 'gvfs-gphoto2-volume-monitor'
[ DEBUG   ] pibooth           : Try to kill process 'gvfsd-gphoto2'
[ DEBUG   ] pibooth           : Found gPhoto2 cameras on ports: 'usb:001,004'
[ INFO    ] pibooth           : Configuring hybrid camera (OpenCV + gPhoto2) ...
[ DEBUG   ] pibooth           : Preview resolution is (640.0, 480.0)
[ WARNING ] pibooth           : The connected DSLR camera is not compatible with preview
[ DEBUG   ] pibooth           : Setting option imgsettings/iso=100
[ DEBUG   ] pibooth           : Setting option settings/capturetarget=Memory card
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : took 10.434 seconds
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[... 59 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 138, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 102, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 71, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 155, in preview
    self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 101, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ DEBUG   ] pibooth           : took 0.018 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.1.21-v8+', version='#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023', machine='aarch64', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.046 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'

Hopefully someone can make more sense of this than me.

Thanks in advance

ricso, 2024-05-17:
Does

> Hi, @TboneSteak237 as stated in the documentation pibooth doesn't support libcamera yet so if you want to use a picamera you need to use an older OS (Buster).
> 
> See #540 and #526 that follow this need.

Does this mean that Pibooth should work with Bookworm if another external camera is used (DSLR or Webcam)?

werdeil, 2024-05-17:
Pibooth could work with external camera and Bookworm, from some of the post here it looks like it but I haven't tested it myself.

ricso, 2024-05-17:
Ok. Thanks for the quick reply!
I'm going to try and I'll report back