pibooth/pibooth #567: Camera works with Gphoto2 but not with pibooth :(

niccolov, 2024-04-02:
I'm trying to set up pibooth on a Raspberry 3 b+.

I managed to install everything and the environment seems to work fine.
I'm trying to use the system with a Nikon Coolpix S33 camera and a Canon Selphy CP910 printer.

Everything seems to work with gphoto2 and cups, but when I launch pibooth the camera stops working.

If I use the Nikon Coolpix S33 with gphoto2 I can take photos without problems.

Here are the results:

gphoto2 --capture-image-and-download --filename=test-2.jpg

Il nuovo file è alla posizione /store_00010001/DCIM/146NIKON/DSCN3059.JPG sulla fotocamera
File salvato come "test-2.jpg"
Eliminazione del file /store_00010001/DCIM/146NIKON/DSCN3059.JPG sulla fotocamera

However, if I launch the Pibooth program I always get the following error:

[ DEBUG   ] pibooth           : took 4.147 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : [-6] Unsupported operation
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 98, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 501, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 119, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 473, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 99, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 61, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 470, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 138, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 102, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 71, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/gphoto.py", line 220, in preview
  self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/gphoto.py", line 129, in _get_preview_image
    cam_file = self._cam.capture_preview()
gphoto2.GPhoto2Error: [-6] Unsupported operation
[ DEBUG   ] pibooth           : took 0.006 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='iltuophotoboot', release='6.1.21-v7+', version='#1642 SMP Mon Apr  3 17:20:52 BST 2023', machine='armv7l', processor=''

If I use the command pibooth-diag I get the following message:

pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
Pibooth version installed: 2.0.8
Installed plugins: google-photo-1.2.3
GPhoto2 version installed: 2.5.27
  - standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
  - gcc (C compiler used)
  - ltdl (for portable loading of camlibs)
  - EXIF (for special handling of EXIF files)
Listing all connected DSLR camera
00 : addr-> usb:001,009  name-> USB PTP Class Camera

Stating diagnostic of connected DSLR camera
Gphoto2: gp_port_set_error [gphoto2-port.c:1190]: The supplied vendor or product id (0x0,0x0) is not valid.

... details about the camera ....

================================================================================
Testing commands used by pibooth
Setting option imgsettings/iso="100"
   -> unsupported setting imgsettings/iso=100 (nothing configured on DSLR)
Setting option settings/capturetarget="Memory card"
   -> unsupported setting settings/capturetarget=Memory card (nothing configured on DSLR)
Unknown option actions/viewfinder
Take capture preview
Gphoto2: gp_camera_capture_preview [gphoto2-camera.c:1403]: 'camera->functions->capture_preview ( camera, file, context)' failed: -6

================================================================================
ABORT   : exception occures: [-6] Unsupported operation
If you are investigating why pibooth does not work with your DSLR camera,
please paste the content of generated file 'diagnostic.log'
on https://github.com/pibooth/pibooth/issues

What could it depend on? How can I make pibooth work with the Nikon Coolpix S33?

**Environment**
 - OS: Raspbian GNU/Linux 11 (bullseye) 
 - Camera: Nikon Coolpix S33
 - Pibooth version: 2.0.8
 - Gphoto version: 2.5.27

How Can I solve it to make it working?

fabienmaquin, 2024-12-30:
Same over here... Gphoto --capture-image working well and yet gphoto2 seems not to work with pibooth

Rouliooooo, 2025-02-22:
Same here as well, seems like many people get the same issue for months but no patch have been found... So sad, I came to Pibooth precisely to be able to use my DSLR camera..
Already searched into versions compatibility between dependencies but I'm afraid that I don't have enough skills to know if this is the right thing to look at.
Would love someone to lean over
gPhoto2 working well by nature, but Pibooth seems to not find it installed.

tonarec, 2025-05-15:
Hi, I've tweaked a lot the distro to make it work properly with DSLR (Canon 2000D). Here are some tips I can give you that can help:
- Check that the distro is Raspbian Buster
- Make sure you have the latest stable version of gphoto2, libgphoto2, and pibooth is installed with the [dslr] flag: [Manual procedure](https://documentation.pibooth.org/en/stable/sources/install.html#manual-procedure), [gphoto2-updater](https://github.com/gonzalo/gphoto2-updater)
- ISO have to be set to `['Auto']` in `pibooth.cfg` and camera is in auto mode
- You can try to find if the USB is taken by another process rather than pibooth: `sudo fuser -v /dev/bus/usb/*/*`
- Most of the time, gvfs is taking the USB over pibooth, test to kill the process (if exist): `pkill gvfs-gphoto2-volume`
- It's possible the DSLR is blocked, try to unplug/plug the cable or: `sudo usbreset /dev/bus/usb/xxx/xxx`

Hope it helps!