pibooth/pibooth #568: Pi Camera v3 is supported?

niccolov, 2024-04-03:
I've just bought the Pi Camera V3, and waiting for it.... but now I've read an old message saying that Pi Camera V3 is not supported.
Do you confirm it? Do I'have to change it? Does the V2 works fine?

Thanks,

Niccolò

turriert, 2024-04-03:
Hello, 

Picamera v3 is not supported. I have the same problem as you. So I put the picamera v2. It works well, but the photo quality is a little poorer. You still have to wait for compatibility with v3.

Thibaut

werdeil, 2024-04-03:
Hi @niccolov, as already answered by @turriert V3 needs at least the latest OS which is unfortunately not the case yet. I was hoping to find some time to work on that but family priorities came before. Follow the linked topics to get more infos.

FlyJJ1010, 2025-02-16:
Hello, i am working on my photobooth, with Pibooth.
But as you already might guess, i bought the camera v3 😁 to find out its not supported yet 😁
Is there something i can do to help to make this camera supported bij Pibooth?
Or can you maybe share some expected timelines? If its all uncertain i need to switch to the v2 model.
 Thanks

Jbuchs, 2025-04-01:
Hi, I have the exact same issue. I bought the V3 for Pibooth without thinking it could not work. Still no solution for this ?