pibooth/pibooth #571: Mirror photos

BramWerbrouck, 2024-04-07:
Al the pictures taken with my pi camera v2 are mirrored (you can’t read the text on my shirt) how can I solve this problem?

werdeil, 2024-04-08:
Hi @BramWerbrouck, have a look at the `flip` setting in the camera section: 
```
# Flip horizontally the capture
flip = False
```

BramWerbrouck, 2024-04-08:
These lines where missing in my config-file, thanks for this.

The problem now is that the 'live video' just before the caputer is mirrored (just like it was before).. the processed image is correct now (thanks to the flip option in the config-file) can this problem be solved too?

BramWerbrouck, 2024-05-26:
I found the problem. I changed rule 53 from rpi.py from
`def preview(self, window, flip=True): `
into:
`def preview(self, window, flip=False): `

and the problem is solved :-) I think this variable should come from the Config file too, but no id how to do this (het)