pibooth/pibooth #572: Raspberry Pi 4 model B : pas de lancement

lm699, 2024-04-07:
Après avoir fraichement installé tous les pré requis, il n'y a pas de lancement de pibooth.
J'ai une Raspberry Pi 4 model B
Une caméra compatible avec opencv (mais je le l'ai pas installé sur cette version, car j'ai déjà rencontré des bugs et apparement, après buster, il y a des incompatibilités) ;
gphoto2 pour une prise de vue avec un reflex numérique (nikon).
Je n'ai pas installé cups

**Logs**
1. Set the `debug` mode in the config
2. Post your config file 
[config.txt](https://github.com/pibooth/pibooth/files/14898583/config.txt)

lucas@pbooth:~ $ pibooth
pygame 2.1.2 (SDL 2.26.5, Python 3.11.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : +< Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756

4. Post the `diagnostic.log` file if you are using a DSLR camera (generated by running the command `[diag.txt](https://github.com/pibooth/pibooth/files/14898574/diag.txt)`)

**Environment**
 - OS Raspberry Pi OS (cat /etc/os-release)
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
 - Camera Nikon D5300
 
 - Pibooth version
pygame-menu 4.0.7
pygame 2.1.2 (SDL 2.26.5, Python 3.11.2)
2.0.8

Merci d'avance pour votre aide :)

IroMain327, 2024-04-28:
as tu essayé avec une ancienne version de raspbian ? blue eyes ?

lm699, 2024-04-28:
Oui, avec raspbian os ça fonctionne bien.
À la prise de photo, c'est un peu long pour passer à la suite

werdeil, 2024-05-17:
Hello @lm699, not sure to understand your issue, from what I understand it works as expected on Buster.

werdeil, 2025-03-24:
Closing this issue as no answer for a long time, feel free to reopen if you come back.

closed 2025-03-24 by werdeil