pibooth/pibooth #574: Can't view photos

Refurb, 2024-04-12:
Hi all,

My Raspberry and DSLR are in a photobooth I've made which is closed with screws. I want to check the camera device for the original pictures after it is used.

Whenever the Raspberry starts, I see my Canon as a device on the desktop. Normally you can access the media in the explorer/browser of Raspberry. However, as soon as Pibooth runs, the device will be hidden. And is inaccessible. My pibooth always runs automatically whenever my Raspberry starts.

I've tried a lot of commands like killing pibooth and gphoto but nothing happens. I just want to see and delete the pictures on the DSLR through my Raspberry (which I control via Realvnc) . What can I do to release the use of gphoto/pibooth, other then disabling the autostart function of Pibooth, reboot the entire raspberry and control my DSLR and turn the autostart on again for the next time?

The issue also occurs when I use a Nikon DSLR, so I really think pibooth/gphoto are the main reason why I can't do it.

Refurb, 2024-04-25:
I have the idea that pibooth is at the end of it's life? There are no reactions anymore on new topics here lately and I wonder if there ever will be a new update..

moskate49, 2024-04-25:
Hello in my Photobooth i have connected inside au mini keyboard (r8mini) just press echap when on pibooth to go on raspberry desktop. For the photo I have added a usb extension who is under the pibooth . When I whant to see the photo I press echap with the keyboard and copy on a usb key connected on the extension , or on the desktop launch command: pibooth —config and change the whay for saving picture directly on a usb key plugged in the usb extension.