pibooth/pibooth #576: Led start button photo

NiraCity, 2024-04-21:
Hello,

I have a question, is there a way to make the output pin 7 (picture button led) only high or low? For example high if the button is pressed. Now it's flashing but for my setup I need an high or low output from the RPI.

NiraCity, 2024-04-21:
I was playing around with some lines in the light_plugin.py and found out that if you change line 23 from the light_plugin.py to app.leds.on() this changed the behaviour of the led, so the led is not blinking anymore. Exactly what I needed.

DJ-Dingo, 2024-04-21:
@NiraCity 
You can add high or low, in the "pibooth-extra-lights" plugin 
(If i understand your question correct)
https://github.com/pibooth/pibooth-extra-lights