pibooth/pibooth #578: Add possibility for arrow location on the side instead of top or bottom

schorr31, 2024-04-28:
**Is your feature request related to a problem? Please describe.**
I build a photobooth with two buttons which are located on either side of the display. The problem i am facing now is that the arrows can not be pointed towards the side instead of bottom or top.

**Describe the solution you'd like**
A new configuration possibility for pointing the arrows to the side.

**Describe alternatives you've considered**
I tried without the arrows but it is not really clear sometimes what button to press.

Is there an easy fix or is a rewrite of background.py needed?

[image]

werdeil, 2024-05-14:
Hi @schorr31, It needs quite some rewrite in background.py to make it work (all the resize methods) but it can be a nice enhancement to have a new arrow location SIDE to handle your case.

I'm keeping the issue opened, anyone can open a pull request to add it.

DJ-Dingo, 2024-05-17:
Hey @schorr31 
Yes you can do that, "if you manual change the original arrow.png" image inside the asset folder in PiBooth install folder.
(You go to your Python folder and then go to "/pibooth/pictures/assets/" And owerwrite the arrow.png with a new one that is turned 90 degrees (Look at the zip file i prowided, to get you started, you can use Gimp to make new ones that fit your needs. 
 
[arrow.png.zip](https://github.com/pibooth/pibooth/files/15349917/arrow.png.zip)
)

Your PiBooth is installed in your Python folder in my cases  /home/YOURNAME/ then in /.local/lib/python3.9/sitepackages/
And then /pibooth/pictures/assets/

In my furture setup I use a foot pedal, so i chaged the arrow with a foot pedal "and some text obove too.

You have to be aware that when you do this, an reinstall or update Pibooth later, then you have to do it again. So keep a copy of your new arrow.png file / setup somewhere else too.

Here are some test image of how it can look like if you do it. 
You can make the size of the arrow image larger or smaller and make some empty space "Above or Under" the "arrow in the image" to make it go lower or higher on the screen. The rest of align placing you can do in the config file for pybooth (I have used '0' in my pibooth config.cfg for theese test setups.

[image]
[image]
[image]
[image]
[image]
[image]
[image]