pibooth/pibooth #579: final image took around 2 seconds after the "SMILE" image

miche91, 2024-05-04:
**Describe the bug**
Raspberry 4 and logitec c270 (or trust webcam)
Preview looks ok without delay,  the "smile" froze frame is perfectly real time but the saved picture is captured 2-3 seconds AFTER. Since the "smile" froze frame is perfect, you think all good so you walk away and the saved picture capture you walking away becasue there is a delay.
facts: 640x480 delay= 1.5sec , 1280x720 delay=2.5 sec
lowering down preview resolution does not have any impact (tried to tweak it down to 320x240 in opencv.py)

tried different settings , to restore settings to default, to reinstall OS and to reinstall pibooth.
tried 2 different cameras, same issue 

If I take a photo from command line with the same camera at ANY resolution there is absolutely no delay and it is perfect.
I am not sure it as a problem related to the preview since changing preview resolution does not make any impact.
I am confused since it looks reducing the final image resolution HAVE an impact 

Thanks in advance for the help
I can post logs and config if necessary but they are inconclusive to me since I only see that the "Capture" phase is like 3 seconds

miche91, 2024-05-07:
anyone with the same issue?

miche91, 2024-05-09:
tried on a laptop and issue is the same with 3 USB webcams 

I believe this is a pibooth limitation

chriskuipers, 2024-05-12:
Yes, I've got the same issue although the delay is less (using an old iMac). Other thing I've noticed is that while the lighting during the preview is great, the photos turn out pretty dark. But I'll raise a new issue about that so not to clutter your issue 😊

miche91, 2024-05-13:
hey @chriskuipers 
no worries, but I believe nobody is supporting this app aymore. It has probably been kinda abandoned

werdeil, 2024-05-14:
Hi @miche91, even if we are less active we are still around to help. Can you post the logs?

Your configuration (an external webcam) is not the one for which pibooth was originally coded (pi camera or hybrid picamera/ DSLR). The way it is coded right now is that in the case we have only one source we stop the preview to take the photo and put back the preview after. In your configuration it seems to take too much time, to be honest I'm not sure we can do something about it without having the same setup.

miche91, 2024-05-14:
thanks @werdeil 
here is my config and logs, from my ubuntu laptop using the logitech USB webcam (same behavior on the raspberry)

thanks 

[pibooth.zip](https://github.com/pibooth/pibooth/files/15308588/pibooth.zip)

jsmrepairs, 2024-08-02:
@miche91 Did you get this issue solved? I'm having the same trouble

rogusdev, 2024-12-31:
I am also having this issue. I am using an external webcam, and it is about 2 sec delay after smile until photo is taken. I looked at the code and as @werdeil says: https://github.com/pibooth/pibooth/blob/b05f1ec53b092ef4bdf7b3600c92b41581ca9424/pibooth/camera/opencv.py#L157 `preview_countdown` finishes with a static preview image, and then https://github.com/pibooth/pibooth/blob/b05f1ec53b092ef4bdf7b3600c92b41581ca9424/pibooth/camera/opencv.py#L204 `capture` has to go through a bunch of steps with no preview showing while it collects all the information before finally taking the photo. I suspect that the only way this could be avoided would be to have a separate thread collecting the final photo, while the preview continued showing, but I don't know if the camera would allow this.