pibooth/pibooth #580: Printer not working

NiraCity, 2024-05-08:
Hallo evryone.

I'm almost done with the photoboorh, but I can't print yet.

I have an HP LaserJet Pro M255. 
And I can make a test print from cups. (localhost:631)

Only with Pibooth it doesn't  print photos.

In the config file I have (under printer)

leave the name as default. but even when I enter the printer name there, it does not print.

printer_option is empty

printer view 10 sec

number of pages automatically sent to the printer = 1

Maximum number of printed pages = - 1

Maximum nuber of printed pages = 3

picture per page is 1.

Does anyone know what else I need to adjust to get the printer working?

NiraCity, 2024-05-13:
$ pibooth-printcfg --json
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
  return f(*args, **kwds)
pygame 2.5.2 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
^[[A^[[AINFO:pibooth:Connected to printer 'HP_ColorLaserJet_M255-M256'
{"PageSize": "A4", "PageRegion": "A4", "Duplex": "None", "InputSlot": "Auto", "HPOption_Duplexer": "False", "HPPJLColorAsGray": "no", "HPRGBEmulation": "DefaultSRGB", "HPEdgeControl": "Normal", "HPGeneralHalftone": "Smooth", "HPTextNeutralGrays": "Black", "HPGraphicsNeutralGrays": "Black", "HPPhotoNeutralGrays": "ProcessBlack", "HPPJLEconoMode": "False", "MediaType": "Unspecified"}

 $ pibooth-diag
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
  return f(*args, **kwds)
pygame 2.5.2 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
Pibooth version installed: 2.0.8
Installed plugins: sound-effects-0.0.3, extra-lights-1.0.2, picture-template-1.1.0
gPhoto2 not installed, cannot diagnose connected DSLR

NiraCity, 2024-05-13:
When starting the software i see a messag.
~ $ pibooth
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
  return f(*args, **kwds)
pygame 2.5.2 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: sound-effects-0.0.3, extra-lights-1.0.2, picture-template-1.1.0
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
VIDIOC_QUERYCTRL: Input/output error
[ INFO    ] pibooth           : Configuring OpenCV camera ...
VIDEOIO ERROR: V4L2: setting property #30 is not supported
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)

Anybody know what i can change?

NiraCity, 2024-05-13:
Tomorrow I will try to Install pycups by the follow link

https://github.com/anxuae/pycups-notify.git
And
https://github.com/OpenPrinting/pycups.git

Hope that will work.

werdeil, 2024-05-14:
Hi @NiraCity , you indeed need to have both pycups and pycups-notify installed. Both shall have been installed along with pibooth when running 

```sudo pip3 install pibooth[dslr,printer]```

Did you install with this command? You can always fix it by running the following command:

```sudo pip3 install pycups==1.9.73 pycups-notify==0.0.4```

NiraCity, 2024-05-15:
Thank you very much for your response.

Now it works, so I should have installed an older version?

Thank you

werdeil, 2024-05-17:
> Now it works, so I should have installed an older version?

No I think you installed it without the printer option that's all. I added the documentation tag to improve it on this (not the common pip command to have this kind of option)

closed 2024-05-17 by werdeil