pibooth/pibooth #582: Pibooth 2.0.8 with pi5 et picamV3 - OSError: libbcm_host.so: cannot open shared object file: No such file or directory

Romain34725, 2024-05-13:
**Describe the bug**
when i start pibooth, the program don't start 

**Logs**
pibooth
pygame 2.5.2 (SDL 2.28.3, Python 3.11.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/home/romain/.local/bin/pibooth", line 5, in <module>
    from pibooth.booth import main
  File "/home/romain/.local/lib/python3.11/site-packages/pibooth/booth.py", line 28, in <module>
    from pibooth.plugins import create_plugin_manager
  File "/home/romain/.local/lib/python3.11/site-packages/pibooth/plugins/__init__.py", line 8, in <module>
    from pibooth.plugins.camera_plugin import CameraPlugin
  File "/home/romain/.local/lib/python3.11/site-packages/pibooth/plugins/camera_plugin.py", line 6, in <module>
    from pibooth import camera
  File "/home/romain/.local/lib/python3.11/site-packages/pibooth/camera/__init__.py", line 4, in <module>
    from pibooth.camera.rpi import RpiCamera, get_rpi_camera_proxy
  File "/home/romain/.local/lib/python3.11/site-packages/pibooth/camera/rpi.py", line 8, in <module>
    import picamera
  File "/home/romain/.local/lib/python3.11/site-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/home/romain/.local/lib/python3.11/site-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/home/romain/.local/lib/python3.11/site-packages/picamera/mmal.py", line 47, in <module>
    from .bcm_host import VCOS_UNSIGNED
  File "/home/romain/.local/lib/python3.11/site-packages/picamera/bcm_host.py", line 47, in <module>
    _lib = ct.CDLL('libbcm_host.so')
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: libbcm_host.so: cannot open shared object file: No such file or directory

**Environment**
 - OS [12 (bookworm)]
 - Camera [picam v3]
 - Pibooth version  [2.0.8]

do you have a solution for this problem ?

Thanks
Romain

werdeil, 2024-05-14:
Hi @Romain34725, unfortunately Pibooth 2.x is not compatible with Raspi OS > Buster.

You can try the v3.0 branch with no garantees as it is not yet finished and not tested on pi5 nor picamv3.

GrigoriOH, 2025-01-20:
Sorry to bother, but is there any roadmap / plan on when to include RaspiOS > Buster in pibooth?