pibooth/pibooth #588: Pibooth not starting - video system not initialised

timoschuett, 2024-06-16:
Hi,
i cant get pibooth to start. 
The installation was step by step followed on the website.

Hardware: Raspberry 3b+, Sony SLT a37

I get the following error:

pygame 2.1.2 (SDL 2.26.5, Python 3.11.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: stillimage-camera-0.1.4
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pibooth/booth.py", line 96, in __init__
    self._window = PiWindow(title, init_size, color=init_color,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pibooth/view/window.py", line 56, in __init__
    info = pygame.display.Info()
           ^^^^^^^^^^^^^^^^^^^^^
pygame.error: video system not initialized

Help is much appreciated.
Thanks,
Timo

werdeil, 2024-06-16:
Hi @timoschuett, can you provide more info on the setup? What kind of screen are you using, how is it wired to the raspberry pi?

karlzre, 2024-11-13:
was too fast, now i enabled the raspebry camera, i get this error:

[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
The path /dev/dri/ cannot be opened or is not available
The path /dev/dri/ cannot be opened or is not available
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 96, in __init__
    self._window = PiWindow(title, init_size, color=init_color,
  File "/usr/local/lib/python3.9/dist-packages/pibooth/view/window.py", line 56, in __init__
    info = pygame.display.Info()
pygame.error: video system not initialized