pibooth/pibooth #590: Pibooth crashes upon launch after I got my Pi4B to recognize my Camera Module v2

JackOnGH, 2024-06-20:
I am a newbie to Pi's and installed pibooth. 
I was able to launch it and get up to the point where it wants to take the first photo, then it said there was a problem.

It seemed that my Pi4B couldn't "see" my connected "Camera Module 2"

Going off of the advice in the latter half of [this YouTube video](https://www.youtube.com/watch?v=Z8cs1cRrc5A), in my config.txt file I changed

**start_x=1** 

to 

**#start_x=1**

and then added the line

**camera_auto_detect=1**

right below that

and changed 

**gpu_mem=128**

to

**gpu_mem=256**

This worked to get my Pi to actually _see_ the camera 
(like, testing with the libcamera-hello command now showed an image, 
where prior to this, my Pi told me there was no camera installed)

But now, when I launch pibooth, it brings up the edges of the window, but it crashes without a message before the program can even display the black background and text.

Is this a known issue?
Is there a simple ("for idiots") fix for this?

I also have a 16MP ArduCam which I would prefer to use, as it would take larger photos and has a good lens for photobooth photos, but I know that pibooth usually is looking for one of the "official" Pi cameras. But again, if there's an easy _"do this, by running the following terminal commands"_ to get pibooth working, and with the _ArduCam_, that would be great.

So, long story short:

I am hoping to get pibooth up and running, at least with the v2 Pi cam, at best with the 16MP ArduCam

werdeil, 2024-06-22:
Hi @JackOnGH, as stated in the doc pibooth with a picamera is not yet compatible with Bullseye OS and above. See #540 for the follow up on this topic.

Closing this issue then.

closed 2024-06-22 by werdeil

JackOnGH, 2024-06-22:
Thank you for letting me know—sorry I missed that.

OS Buster won’t run on my Pi4B, so it just seems that it’s not to be.

Thanks again for letting me know!

-J

werdeil, 2024-06-22:
You can try the v3 branch (https://github.com/pibooth/pibooth/tree/3.x), we worked on it a few time ago but never found time to test it enough.

JackOnGH, 2024-06-22:
Thanks for the info.

I was trying to set up a *Traveling Photo Booth* for an upcoming trip to
Nepal,
but I’ve already *twice*(!) ordered and returned a
Pi4B, microSD, external screen, cables, and Pi cam and ArduCam,
so I think Amazon may be tiring of my shenanigans.

But, in case I *do* try one more time, are you saying,
use this 3.x version with a *Pi4B* again?
Or do I need to use *an earlier Pi* like a Pi 3?

Also, from the link on the 3.x page, it looks like the installation steps
are identical...

Can I ask, do you know if it would be easy
(easy like *"just cut-and-paste the following line of text into this part
of the ____.txt file"*)

1. to just have *one* button (I don't want a print-at-the-booth option—I'll
just print everything myself at a photo place) to start the photo-taking.

2. and to only have one option for photo layout (just 4 photos instead of
an option to just take 1) ?

Thanks for your help with this!

- Jack