pibooth/pibooth #594: pibooth crashes after several pictures taken

fbourgeois24, 2024-07-12:
**Describe the bug**
I can take several pictures but after some time (random number of pictures taken) pibooth crashes on the "smile" screen.
When the bug appends, the activity led on the camera still light (it turns off usually)
I already tried to decrease camera definition but the bug persists.

**Logs**
Config file
```
[GENERAL]
# User interface language: 'br', 'de', 'dk', 'en', 'es', 'fr', 'hu', 'it', 'nl', 'no' or 'pt'
language = fr

# Path to save pictures (list of quoted paths accepted)
directory = /home/pi/Pictures/pibooth

# Start pibooth at Raspberry Pi startup
autostart = True

# How long to wait in seconds before start pibooth at Raspberry Pi startup
autostart_delay = 5

# In debug mode, exceptions are not caught, logs are more verbose, pictures are cleared at startup
debug = True

# Path to custom plugin(s) not installed with pip (list of quoted paths accepted)
plugins = 

# Plugin names to be disabled after startup (list of quoted names accepted)
plugins_disabled = 

# Enable a virtual keyboard in the settings interface
vkeyboard = False

[WINDOW]
# The (width, height) of the display window or 'fullscreen'
size = fullscreen

# Background RGB color or image path
background = (0, 0, 200)

# Font name or file path used for app texts
font = Amatic-Bold

# Text RGB color
text_color = (255, 255, 255)

# Blinking background when a capture is taken
flash = True

# Animate the last taken picture by displaying captures one by one
animate = False

# How long is displayed the capture in seconds before switching to the next one
animate_delay = 0.2

# On 'finish' state: how long is displayed the final picture in seconds (0 if never shown)
finish_picture_delay = 10

# On 'wait' state: how long is displayed the final picture in seconds before being hidden (-1 if never hidden)
wait_picture_delay = 0

# How long is displayed the 'chosen' state:  (0 if never shown)
chosen_delay = 0

# Show arrows to indicate physical buttons: 'bottom', 'top', 'hidden' or 'touchscreen'
arrows = touchscreen

# Apply horizontal offset to arrows position
arrows_x_offset = 0

[... 555 lines cut ...]
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 415, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 116, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 80, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/camera_plugin.py", line 103, in state_capture_do
    app.camera.capture(effect)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/camera/opencv.py", line 221, in capture
    ret, image = self._cam.read()
cv2.error: /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/imgcodecs/src/loadsave.cpp:637: error: (-215) !buf.empty() && buf.isContinuous() in function imdecode_

[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.103-v7l+', version='#1529 SMP Tue Mar 8 12:24:00 GMT 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
```

**Environment**
- Raspberry pi 4 
 - OS Raspbian GNU/Linux 10 (buster)
 - Camera Logitech BRIO 4K Stream edition
 - Pibooth version 2.0.8 (using opencv, not installed gphoto2 or cups)

Exitras, 2024-07-13:
I experience the same. Pibooth randomly crashes after some time.

valentin-harrang, 2024-09-02:
@fbourgeois24 Have you found a solution?

SebastiaanRobbrecht, 2024-10-29:
Same problem here but with a Raspberry 3B