pibooth/pibooth #607: Delay on preview with Canon 1200D

adrien35136, 2024-09-02:
Hello, 

Thanks for your work. Everything is working well except with the preview of the camera. When i'm moving there is a delay of 2 or 3 seconds with the preview on my touchscreen. I tried with picamera V2 and there is no delay. 

Do you know why have a delay with my camera CANON 1200D ?

Setup:
Raspberry pi 3 B+
Pibooth version : 2.08 or 2.07

Thanks in advance
Adrien

werdeil, 2024-09-02:
Hello @adrien35136, 

This behavior is "normal", when the preview uses the DSLR it is taking capture continuously so it is slower than the preview on the picamera where it directly show the picamera view above the application. That's why we advise to use the hybrid mode (use the picamera for the preview and the DSLR for final picture).

adrien35136, 2024-09-02:
Thank you for your answer. Can you tell me how I can do the hybrid mode ? Is it possible to have a full screen preview with the picamera instead of only a rectangle ? 

Thanks in advance 
Adrien

werdeil, 2024-09-03:
Hybrid mode is automatically enabled when picamera is plugged and enabled and gphoto is installed with a DSLR plugged.

For the preview it fits the size of the photo in the final picture, going fullscreen will make it cropped after.

adrien35136, 2024-09-05:
Hello,

Thank you for your answer. It works for me in hybrid mode but only with
this resolution (resolution = (1934, 2464)) but I have warnings and a minor
error with the value of iso (see logs in attached). I don't know if it's
normal to have theses warnings

When i'm trying to increase the resolution (full resolution = 3280, 2464) i
have an error "out of resources" (see logs in attached). I made it work by
increasing the GPU memory from 128 to 256 in raspberry-config.

Then i installed this plugins to have a generic template :

$ pip3 install pibooth-picture-template

Since i installed this plugin, pibooth crash during the display of the
4 pictures so I need to power the raspberry off. It works when i'm
taking only one picture. I think there is a problem with
multiprocessing (see log in attached)

Do you have advice for me ?

Setup :
Raspberry pi 3 B+
Picamera v2
Canon 1200D

Thanks in advance,
Adrien

***@***.***:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: picture-template-1.1.0, getready_text-0.0.2
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Failed to allocate required memory.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO    ] pibooth           : Configuring hybrid camera (Picamera + gPhoto2) ...
[ WARNING ] pibooth           : Invalid value '100' for option iso (possible choices: ['Auto']), trying to set it anyway
[ ERROR   ] pibooth           : Unsupported option imgsettings/iso=100 ([-110] I/O in progress), configure your DSLR manually
[ INFO    ] pibooth           : Connected to printer 'Adobe_PDF_via_VNC_de_DESKTOP-N9JN9BS'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ INFO    ] pibooth           : Saving raw captures
[ INFO    ] pibooth           : Creating the final picture
[ INFO    ] pibooth           : Parsing pictures template file: /home/adrien/.config/pibooth/pibooth.xml
[ INFO    ] pibooth           : Found template 'Page-1': 1 captures - 2 texts - 0 others
[ INFO    ] pibooth           : Found template 'Page-2': 2 captures - 2 texts - 0 others
[ INFO    ] pibooth           : Found template 'Page-3': 3 captures - 2 texts - 0 others
[ INFO    ] pibooth           : Found template 'Page-4': 4 captures - 2 texts - 0 others
[ INFO    ] pibooth           : Found template 'Page-5': 1 captures - 2 texts - 0 others
[ INFO    ] pibooth           : Found template 'Page-6': 2 captures - 2 texts - 0 others
[ INFO    ] pibooth           : Found template 'Page-7': 3 captures - 2 texts - 0 others
[ INFO    ] pibooth           : Found template 'Page-8': 4 captures - 2 texts - 0 others
[ INFO    ] pibooth           : Use TemplatePictureFactory to create background
[ INFO    ] pibooth           : Use TemplatePictureFactory to concatenate images
[ INFO    ] pibooth           : Use TemplatePictureFactory to assemble final image
[ INFO    ] pibooth           : Use TemplatePictureFactory to draw texts
[ INFO    ] pibooth           : Save image '/media/adrien/PHOTOMATON/2024-09-05-20-47-41_pibooth.jpg'
[ INFO    ] pibooth           : Display the final picture
[ INFO    ] pibooth           : Moving the picture in the forget folder
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[... 65 lines cut ...]
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.9/multiprocessing/pool.py", line 136, in worker
    put((job, i, (False, wrapped)))
  File "/usr/lib/python3.9/multiprocessing/pool.py", line 136, in worker
    put((job, i, (False, wrapped)))
  File "/usr/lib/python3.9/multiprocessing/queues.py", line 378, in put
    self._writer.send_bytes(obj)
  File "/usr/lib/python3.9/multiprocessing/queues.py", line 378, in put
    self._writer.send_bytes(obj)
  File "/usr/lib/python3.9/multiprocessing/connection.py", line 205, in send_bytes
    self._send_bytes(m[offset:offset + size])
  File "/usr/lib/python3.9/multiprocessing/connection.py", line 205, in send_bytes
    self._send_bytes(m[offset:offset + size])
  File "/usr/lib/python3.9/multiprocessing/connection.py", line 416, in _send_bytes
    self._send(header + buf)
  File "/usr/lib/python3.9/multiprocessing/connection.py", line 416, in _send_bytes
    self._send(header + buf)
  File "/usr/lib/python3.9/multiprocessing/connection.py", line 373, in _send
    n = write(self._handle, buf)
  File "/usr/lib/python3.9/multiprocessing/connection.py", line 373, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
BrokenPipeError: [Errno 32] Broken pipe
/usr/lib/python3.9/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 6 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '