pibooth/pibooth #608: Broken pipe

adrien35136, 2024-09-09:
Hi, 
Pibooth failed sometimes and i don't understand why. Looks like there is a problem with multiprocessing
Can you help me ? 
You will find in attached the log
[log_failed_broken_pipre.txt](https://github.com/user-attachments/files/16934915/log_failed_broken_pipre.txt)

Thanks in advance,
Adrien

werdeil, 2024-09-09:
`Asyncronously generate pictures for animation`

Do you confirm that you enabled the animation? Can you test without it to see if it failed?

adrien35136, 2024-09-13:
Yes animation was enabled. Inded, when i disable it, everything is working fine. Can you tell me how can i fix it ? 
Thanks in advance
Adrien

dominiquegrota, 2024-09-13:
Hi, 
I had the same problem with a raspberry 3A+ (only 512 MB ram) I suppose it is because lack of ram, because the same install on a raspberry 3B+ (1 Go Ram), the animation enabled is working...

adrien35136, 2024-09-13:
Hi,
Thank you for your answer. I will try with a raspberry pi 4.

adrien35136, 2024-10-17:
Hi , i'm back because i got a problem to make pibooth worked with my raspberry pi 4. My Picamera V2 is not detected. I did the same work than with my rapsberry pi 3 B+, so i don't understand why. 
When i enabled the legacy camera my rapsberry is not working at all. 
Can you help me please ? 
Thanks in advance