pibooth/pibooth #609: HQ camera Sony IMX477 does not work on raspi pi 5

stefan-kasch, 2024-09-16:
After some tests, I can use the HQ Camera in full resolution (4056 x 3040 pixels), by increasing the GPU memory to 256Mb.

```bash
sudo raspi-config
```

Go to `Performance Options` > `GPU Memory` > `256`
Then reboot the Raspberry-Pi.

Can you try and give me your feedback?

_Originally posted by @anxuae in https://github.com/pibooth/pibooth/issues/167#issuecomment-855442407_
           

Hi,

i tried to use the hq camera on my raspi 5 but i does not work on pibooth. Trying to increas the GPU memory as suggested in issue  167# does not work on the pi 5 since there is not such an option in the performance options.

On pibooth I got the following error code after trying to take a picture:
[ ERROR   ] pibooth           : Can not get camera preview image
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='6.6.47+rpt-rpi-2712', version='#1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02)', machine='aarch64', processor=''

Generally the camera works on the pi 5. libcamera-hello gives a clear image.

lxne, 2024-10-10:
This worked for me too.

lxne, 2024-10-10:
Are you using Buster or a newer raspiOS?

alexismorsilli, 2024-11-15:
Bonjour j'ai le même problème avec la même configuration raspi 5 8gigas os bookworm. Avez vous trouvé la solution ?

alexismorsilli, 2024-11-19:
@stefan-kasch  it works with this plugin ;)  but the plugin use cpu memory instead of gpu for now so this slows down the camera preview
 you have to wait for an update coming soon . 

https://github.com/Smatnemo/pibooth-picamera2.git