pibooth/pibooth #610: replace footer_text2 by logo

desertseb, 2024-09-18:
Hello,
I would like to replace the 'footer_text2' area with a logo. Is that possible, and how can I do it?

desertseb, 2024-09-18:
How can I add a background file?

desertseb, 2024-09-18:
I modified the background field in the PICTURE section of pibooth.cfg by setting backgrounds = background.png, and I get a feedback/error during the photo assembly.

dominiquegrota, 2024-09-18:
Hi, 
Either you have to give the complete link to your "background.png"   (maybe : /home/pi/Pictures/background.png)
or you got the same problem I had here : https://github.com/pibooth/pibooth/issues/397
to solve it I added the plugin "picture template plugin" and the overlays are working

desertseb, 2024-09-26:
thank you it works

closed 2024-09-26 by desertseb