pibooth/pibooth #612: Final resolution

davtech42, 2024-09-27:
Hello,
Pibooth works very well, but the final resolution is not very good, the images are only 96ppp.
Is it possible to have 300ppp?
I use it with a Nikon D5000.

lxne, 2024-10-09:
Hi, with ppp you probably mean the same thing as dpi.
There is a lot to know about [resolution and dpi](https://blog.ironmarkusa.com/dpi-resolution).

In short, the resolution in combination with dpi results in an image printed at a specific size (if the printer respects the dpi settings). Changing the dpi doesn't change the digital resolution of the file. You can have a file with 3000x2000pixel with 96dpi and another one with 3000x2000pixel and 300dpi – the files will have the same 'quality' but the one with 300dpi will be printed much smaller so that you get the 300 dots per inch on paper. If you print it bigger, the dpi will be reduced.

For pibooth, did you set the native resolution of your camera in the config?
If there is some other value, your images will be scaled and loose 'quality'.
pibooth --config

davtech42, 2024-10-22:
Hello
Thank you for your reply.
I did set the native resolution of my camera (4288*2848), but I think the printing of my photos could be better if the final image was in 300dpi.
On the camera's memory card it's 300dpi and in the RAW folder it's 96dpi...
ps: sorry for my english, I use a translator.