pibooth/pibooth #617: Question

acatoire, 2024-10-25:
Any reason why Pillow and pygame-menu requirement are fixed?
'Pillow==9.2.0',
'pygame-menu==4.0.7',

Should we make test to be able to bump the version?

werdeil, 2024-10-25:
Hi @acatoire,

Pillow is fixed because the new version (Pillow 10.+ ) are not working and we wanted to bump the version while going to Pibooth v3.

Pygame-menu is quite the same, we had issues at one time with version 4.1 and didn't take the time to see if the issue is still present on the new versions.

So in a nutshell: we know that the next version was generating issues so we fixed the version. The main idea was to wait for the v3 to clean things but we are too busy to finish it at the moment.

werdeil, 2024-10-25:
If as a QA you want to do some tests feel free to test the v3 of Pibooth to clarify if it is stable enough to be released.