pibooth/pibooth #618: Installation Issues. I can't get pi booth to work on a raspberry pi 4

SebastiaanRobbrecht, 2024-10-29:
**I am trying to install Pibooth on my Raspberry PI 4. But I can't get it to work. I tried the manual installation, I also tried Tijof playbook. I tried different operating systems. 2 different SD cards. Nothing helps...** 

```
software@raspberrypi:~ $ sudo pip3 install pibooth[dslr,printer]
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv
```

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

**I then tried it through the virtual environment. The installation succeeded. But then came the next problem:**

```
(venv) software@raspberrypi:~/pibooth_env $ touch ~/.config/pibooth/pibooth.log
pibooth
<frozen importlib._bootstrap>:241: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
pygame 2.6.1 (SDL 2.26.5, Python 3.11.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
/home/software/pibooth_env/venv/lib/python3.11/site-packages/gpiozero/devices.py:297: NativePinFactoryFallback: Falling back to the experimental pin factory NativeFactory because no other pin factory could be loaded. For best results, install RPi.GPIO or pigpio. See https://gpiozero.readthedocs.io/en/stable/api_pins.html for more information.
  warnings.warn(NativePinFactoryFallback(native_fallback_message))
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
/home/software/pibooth_env/venv/lib/python3.11/site-packages/pluggy/_manager.py:476: PluggyTeardownRaisedWarning: A plugin raised an exception during an old-style hookwrapper teardown.
Plugin: pibooth-core:camera, Hook: pibooth_setup_camera
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected
For more information see https://pluggy.readthedocs.io/en/stable/api_reference.html#pluggy.PluggyTeardownRaisedWarning
  outcome = Result.from_call(
Traceback (most recent call last):
  File "/home/software/pibooth_env/venv/bin/pibooth", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pibooth/booth.py", line 131, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pluggy/_hooks.py", line 513, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pluggy/_manager.py", line 120, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pluggy/_manager.py", line 480, in traced_hookexec
    return outcome.get_result()
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pluggy/_result.py", line 100, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pluggy/_result.py", line 62, in from_call
    result = func()
             ^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pluggy/_manager.py", line 477, in <lambda>
    lambda: oldcall(hook_name, hook_impls, caller_kwargs, firstresult)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pluggy/_callers.py", line 156, in _multicall
    teardown[0].send(outcome)
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pibooth/plugins/camera_plugin.py", line 28, in pibooth_setup_camera
    cam = camera.find_camera()
          ^^^^^^^^^^^^^^^^^^^^
  File "/home/software/pibooth_env/venv/lib/python3.11/site-packages/pibooth/camera/__init__.py", line 52, in find_camera
    raise EnvironmentError("Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected")
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected
(venv) software@raspberrypi:~/pibooth_env $

```
**After testing, Gphoto2 appears to recognize my camera Nikon D3100. And also a test shot succeeds.**

mikehooper, 2024-10-31:
Which version of Raspberry Pi OS are you using? I am using Buster as per the requirements.

I find that if the desktop or file manager has auto started gphoto2, I have to kill it before running pibooth.

`pkill -f gphoto2`

SebastiaanRobbrecht, 2024-11-03:
I'm using a  model Raspberry Pi 4 Model B Rev 1.2
Operating System: Raspbian GNU/Linux 12 (bookworm)

SebastiaanRobbrecht, 2024-11-05:
Tryed installing Buster: 
`software@raspberrypi:~ $ pibooth
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
  return f(*args, **kwds)
pygame 2.6.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Generate the auto-startup file in '/home/software/.config/autostart'
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 97, in __init__
    text_color=init_text_color, debug=init_debug)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/view/window.py", line 61, in __init__
    self.surface = pygame.display.set_mode(self.__size, pygame.RESIZABLE)
pygame.error: Could not initialize EGL`

mikehooper, 2024-11-08:
Did you use Buster with Desktop?

Are you connected to an hdmi monitor?

SebastiaanRobbrecht, 2024-11-08:
Yes buster with Desktop but it's a raspberry pi touchscreen not HDMI.

mikehooper, 2024-11-08:
That should be ok, are you connecting remotely to the Pi using ssh/vnc or have you got a keyboard connected directly to the Pi?

That is a basic pygame error not being able to create a window.

SebastiaanRobbrecht, 2024-11-08:
I use SSH connection, but I have a bluetooth keyboard connected.

mikehooper, 2024-11-08:
Don't use ssh. Use a keyboard directly on the Pi or VNC if you have too.

SebastiaanRobbrecht, 2024-11-08:
Do I have to do a clean install?

mikehooper, 2024-11-08:
No

SebastiaanRobbrecht, 2024-11-09:
We are getting there! When I boot up the pi booth pibooth automatically. However, I still fail to take pictures with my Nikon D3100. No more Nikon D3100 logo appears on the desktop. When I try a test of gphoto2 I get the following message.

```

software@raspberrypi:~ $ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Nikon DSC D3100 (PTP mode)     usb:001,003     
software@raspberrypi:~ $ env LANG=C sudo gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download
                                                                               
*** Error ***              
PTP Operation Not Supported

*** Error ***              
PTP General Error
ERROR: Could not capture image.
ERROR: Could not capture.

```