pibooth/pibooth #621: pibooth is taking 16 pictures instead of 4

SebastiaanRobbrecht, 2024-11-15:
**Describe the bug**
When I start a photo sequence the routine does not do what it is supose to do. 

When I want to take 1 picture the camera takes 4
When I want to take 4 pictures the camera takes 16

the countdown gets stuck at number 3

**Steps to reproduce the bug**

see film

**Logs**
1. Set the `debug` mode in the config
2. Post your config file
3. Copy-paste all the logs from the console
4. Post the `diagnostic.log` file if you are using a DSLR camera (generated by running the command `pibooth-diag`)

**Environment**
 - OS [e.g. Raspbian Buster]
 - Camera [e.g. Rpi camera + Nikon D90]
 - Pibooth version [e.g. 1.1.2]

asheenalovessuzie, 2024-11-17:
Hi, I'm currently facing the same issue, were you able to solve it?

SebastiaanRobbrecht, 2024-11-17:
> Hi, I'm currently facing the same issue, were you able to solve it?

Nope, not yet.

werdeil, 2024-11-18:
Hi can you clarify your issue because there seems to be two different ones:
- From the logs it seems that you manage to get a full workflow (start --> choose --> preview --> capture --> print) is it the case?
- From what I understand you are using pibooth with only a DSLR (Nikon D3100), in this case preview is not optimal as pibooth is creating a preview by doing multiple shots, that may be what you mean by 16 pictures instead of 4.
- About gphoto, did you install it as defined in the [documentation] ?(https://documentation.pibooth.org/en/latest/sources/install.html#install), mainly this part: 
```
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh
wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/.env
chmod +x gphoto2-updater.sh
sudo ./gphoto2-updater.sh 
```

SebastiaanRobbrecht, 2024-11-18:
> Hi can you clarify your issue because there seems to be two different ones:

This is possible...

> 
> * From the logs it seems that you manage to get a full workflow (start --> choose --> preview --> capture --> print) is it the case?

Yes, there is a full workflow

> * From what I understand you are using pibooth with only a DSLR (Nikon D3100), in this case preview is not optimal as pibooth is creating a preview by doing multiple shots, that may be what you mean by 16 pictures instead of 4.

So, I need stop the preview part?

> * About gphoto, did you install it as defined in the [documentation] ?(https://documentation.pibooth.org/en/latest/sources/install.html#install), mainly this part:
>
> ```
> wget [...]

Yes, I did. But at first start I had the problem that my Camera was recognised 2 times by gphoto2. so I was tod to kill gphoto2.

werdeil, 2024-11-18:
To summarize my point about preview with a DSLR, it is expected to hear/see the DSLR do multiple shots in the preview state before the actual capture.
As said pibooth is not capable to doing a better preview with gphoto. The best usage of pibooth's preview is either with the picamera or what we call Hybrid mode (preview done by the picamera and capture by the DSLR).

SebastiaanRobbrecht, 2024-11-18:
> To summarize my point about preview with a DSLR, it is expected to hear/see the DSLR do multiple shots in the preview state before the actual capture. As said pibooth is not capable to doing a [...]

Sorry, I'm a bit of a noob  on the subject. If I dont have a picamera I need to disable preview in pibooth config? Or install a second camera (picamera) for hybrid mode?

werdeil, 2024-11-18:
Either you consider that the preview mechanism is good enough like this (with multiple shots) or you'll need to find another solution (add a picamera, find another tool to do a Photo Booth... ). Unfortunately we didn't manage to do better we gphoto in pibooth.

Is it the only issue you have? Seems that there is something else.

werdeil, 2024-11-20:
> I bought a picamera. How do I set up the hybrid mode?

Enable the camera on your raspberry pi (through raspi-config) and it should be automatically detected by pibooth

> Well some time the sequence does not finish and crashes

Most likely linked (DSLR is too busy to provide a photo for the preview).

SebastiaanRobbrecht, 2024-11-20:
Now it is only de picamera who is taking pictures. gphoto2 states that the Nikon is connected.

xsevcikm, 2024-12-04:
I’m facing a similar issue with Pibooth. Initially, it was capturing four times more pictures than expected, so I added a Raspberry Camera v2 alongside my DSLR (Nikon D3300). Both cameras are detected correctly, and both are capable of capturing images. However, the Raspberry Camera is being prioritized for taking pictures instead of the DSLR, even though I want the DSLR to handle the image capture while using the Raspberry Camera for the preview. How can I ensure the DSLR is prioritized for capturing photos?

hardy-homeautomation, 2025-03-30:
I have the same issue with a Canon EOS 350D - the workflow often carries through but without a countdown, just taking four images about 20 seconds apart.  Occasionally it crashes trying to produce a preview ("Can not get camera preview image").  I have looked through other similar issues here and mostly the answer is switch to a Picamera since this seems to be a common side effect of a DSLR combined with gphoto2.  
I would like to more closely emulate an old-school photo booth where the subject doesn't get to chose whether or not to print and there is no preview.  Ideally this setup won't even use a screen just some led's to signal a photo is about to be taken.
Is it possible for someone to point out where I would start if I wanted to remove preview altogether?  This has also been brought up a few times but I could not find any actual suggestions specific to skipping preview within the code itself.
Thank you!