pibooth/pibooth #622: Unable to run pibooth on my rpi 4 model B

asheenalovessuzie, 2024-11-16:
Hello, I am currently using a raspberry pi 4 model B running on the latest OS Bullseye. I've tried following the steps in the installation guide, and the first issue I faced was in installing pibooth, in which it said that it was externally managed. So, I ended up having to install it on a virtual environment. With only my DSLR camera connected, I tried running pibooth and I get the following:

```
pygame 2.6.1 (SDL 2.28.4, Python 3.11.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/devices.py:297: NativePinFactoryFallback: Falling back to the experimental pin factory NativeFactory because no other pin factory could be loaded. For best results, install RPi.GPIO or pigpio. See https://gpiozero.readthedocs.io/en/stable/api_pins.html for more information.
  warnings.warn(NativePinFactoryFallback(native_fallback_message))
[ INFO    ] pibooth           : Installed plugins:
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
[ WARNING ] pibooth           : Invalid value '100' for option iso (possible choices: ['Auto']), trying to set it anyway
Traceback (most recent call last):
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
Traceback (most recent call last):
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/pibooth/camera/gphoto.py", line 61, in gp_log_callback
    LOGGER.getChild('gphoto2').debug(domain.decode("utf-8") + u': ' + string.decode("utf-8"))
                                     ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'decode'. Did you mean: 'encode'?
[ ERROR   ] pibooth           : Unsupported option imgsettings/iso=100 ([-110] I/O in progress), configure your DSLR manually
Traceback (most recent call last):
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/pins/native.py", line 237, in export
    result = self._exports[pin]
             ~~~~~~~~~~~~~^^^^^
KeyError: 17

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/pins/native.py", line 247, in export
    result = os.open(self.path_value(pin),
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/gpio/gpio17/value'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/aisha/pibooth_env/bin/pibooth", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/pibooth/booth.py", line 133, in __init__
    self.buttons = ButtonBoard(capture="BOARD" + config.get('CONTROLS', 'picture_btn_pin'),
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/boards.py", line 227, in __init__
    **{
      ^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/boards.py", line 228, in <dictcomp>
    name: Button(pin, pull_up=pull_up, active_state=active_state,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/input_devices.py", line 412, in __init__
    super().__init__(
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/mixins.py", line 417, in __init__
    super().__init__(*args, **kwargs)
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/input_devices.py", line 167, in __init__
    self.pin.edges = 'both'
    ^^^^^^^^^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/pins/__init__.py", line 441, in <lambda>
    lambda self, value: self._set_edges(value),
                        ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/pins/native.py", line 519, in _set_edges
    self.factory.fs.export(self._number)
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/pins/native.py", line 251, in export
    with io.open(self.path('export'), 'wb') as f:
OSError: [Errno 22] Invalid argument
Exception ignored in: <function GPIOBase.__del__ at 0x7f7cd4eca0>
Traceback (most recent call last):
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/devices.py", line 153, in __del__
    self.close()
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/mixins.py", line 425, in close
    if self._hold_thread is not None:
       ^^^^^^^^^^^^^^^^^
  File "/home/aisha/pibooth_env/lib/python3.11/site-packages/gpiozero/devices.py", line 448, in __getattr__
    raise AttributeError(f"no such attribute {name}")
AttributeError: no such attribute _hold_thread
```

What does this error mean, is it possible to run pibooth with my current setup?

SebastiaanRobbrecht, 2024-11-17:
I had the same problem. First of all you have to use Raspberry Pi OS Buster (32 bit) with desktop. And are you using SSH?

alexismorsilli, 2024-11-19:
hello same probleme , resolved with : cd /usr/lib/python3.11
sudo rm EXTERNALLY-MANAGED
now work fine :)

asheenalovessuzie, 2024-11-19:
> I had the same problem. First of all you have to use Raspberry Pi OS Buster (32 bit) with desktop. And are you using SSH?

I ended up using Buster instead of Bullseye and it worked fine :) No, I am not using SSH, just a regular monitor and keyboard setup for now.

asheenalovessuzie, 2024-11-19:
> hello same probleme , resolved with : cd /usr/lib/python3.11 sudo rm EXTERNALLY-MANAGED now work fine :)

I'll look into it, thanks!

MahrWe, 2025-02-18:
For me it worked by creating the venv including the dependency packages (e.g. python-opencv) previously installed as per instructions.
Follow installation instructions and before step 8 do `python3 --system-site-packages -m venv  some-dir`
`cd some-dir`
`source bin/activate`