pibooth/pibooth #624: slphy cp1500 with usb

alexismorsilli, 2024-11-30:
Hello, I can't install the selphy cp1500 with usb so that it can work without having an internet connection. With an internet connection no problem it is automatically redesigned in cups, but in usb nothing. if anyone has the solution... Thank you!  gutenprint is already installed .

xsevcikm, 2024-12-04:
Hi, this might help you https://www.youtube.com/watch?v=ZYmsKxRRzVk&t=849s You will need to use a driverless printing setup leveraging the IPP-over-USB protocol.

tonarec, 2025-05-15:
Since pibooth 2.0.8 is stuck with Raspbian Buster, it is also stuck with Gutenprint **5.3.1** (which didn't support the CP1500). To fix that, you have to build Gutenprint **5.3.4** or **5.3.5** (see https://github.com/pibooth/pibooth/issues/476#issuecomment-2056371243).
Here are some good tips from the maintainer to help you build it: https://www.peachyphotos.com/blog/stories/building-modern-gutenprint/
Just make sure to remove the installed Gutenprint, and configure the build to use correct output (`./configure --prefix=/usr`)

NiKoB, 2026-01-11:
@florentin68 , tu parles de Retry mais la politique est à Abort dans ta ligne de commande. Du coup, tu préconise quoi ?