pibooth/pibooth #625: Raspberry camera v2 takes control over DSLR and takes pictures instead

xsevcikm, 2024-12-04:
**Describe the bug**
I am using raspberry pi 4 model b with raspberry pi camera v2 and Nikon D3300 running buster 32 bit. Both are detected and I am able to make pictures with each of them via terminal. When running pibooth, raspberry camera takes over DSLR and preview and picture capturing is done by raspberry camera. When using only DSLR, obviously 4 images are taken, but it works. 

Diagnostic log: for some reason it says gphoto is not installed
[diagnostic.log](https://github.com/user-attachments/files/18011759/diagnostic.log) 

Pibooth config:
[GENERAL]
# User interface language: 'br', 'de', 'dk', 'en', 'es', 'fr', 'hu', 'it', 'nl', 'no' or 'pt'
language = en

# Path to save pictures (list of quoted paths accepted)
directory = ~/Pictures/pibooth

# Start pibooth at Raspberry Pi startup
autostart = True

# How long to wait in seconds before start pibooth at Raspberry Pi startup
autostart_delay = 5

# In debug mode, exceptions are not caught, logs are more verbose, pictures are cleared at startup
debug = True

# Path to custom plugin(s) not installed with pip (list of quoted paths accepted)
plugins = plugins/pibooth_custom_camera.py

# Plugin names to be disabled after startup (list of quoted names accepted)
plugins_disabled = 

# Enable a virtual keyboard in the settings interface
vkeyboard = False

[WINDOW]
# The (width, height) of the display window or 'fullscreen'
size = (800, 480)

# Background RGB color or image path
background = (0, 0, 0)

# Font name or file path used for app texts
font = Amatic-Bold

# Text RGB color
text_color = (255, 255, 255)

# Blinking background when a capture is taken
flash = False

# Animate the last taken picture by displaying captures one by one
animate = True

# How long is displayed the capture in seconds before switching to the next one
animate_delay = 0.2

# On 'finish' state: how long is displayed the final picture in seconds (0 if never shown)
finish_picture_delay = 0

# On 'wait' state: how long is displayed the final picture in seconds before being hidden (-1 if never hidden)
wait_picture_delay = -1

# How long is displayed the 'chosen' state:  (0 if never shown)
chosen_delay = 0

# Show arrows to indicate physical buttons: 'bottom', 'top', 'hidden' or 'touchscreen'
arrows = touchscreen

# Apply horizontal offset to arrows position
arrows_x_offset = 0

# How long is the preview in seconds
[... 252 lines cut ...]
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: |  iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: |  Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : +> took 1.086 seconds
[ DEBUG   ] pibooth           : +< Activate state 'wait'
[ DEBUG   ] pibooth           : |  Add to buffer the image '2127020816'

**Logs**
1. Set the `debug` mode in the config
2. Post your config file
3. Copy-paste all the logs from the console
4. Post the `diagnostic.log` file if you are using a DSLR camera (generated by running the command `pibooth-diag`)

**Environment**
 - OS [e.g. Raspbian Buster]
 - Camera [e.g. Rpi camera + Nikon D90]
 - Pibooth version [e.g. 1.1.2]

Armand-B, 2025-01-10:
Hello, 
I have the same issue. 
Bought a picam v2 because my old DSLR preview wasn't on point, but when I launch Pibooth, everything is handled by the picam.
It's like hybrid mode is disabled or something.
Apart from that, with DSLR alone everything works well and I can preview and take pictures...
Any help please ?

SarkinatorLoramenti, 2025-01-25:
I have the same issue. In the log I get the DEBUG note:

> Fallback to pibooth default camera management system

Here are the two entries in the log for the two cameras.

with PICAM and NIKON D750

> [ INFO    ] pibooth           : Installed plugins: picture-template-1.1.0
> [ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
> [ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
> [ INFO    ] pibooth           : Configuring Picamera camera ...
> [ INFO    ] pibooth           : Connected to printer 'Canon_MB2100_series'
> [ DEBUG   ] pibooth           : Activate state 'wait'

With Nikon D750

> [ INFO    ] pibooth           : Installed plugins: picture-template-1.1.0
> [ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
> [ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
> [ INFO    ] pibooth           : Configuring OpenCV camera ...
> [ DEBUG   ] pibooth           : Preview resolution is (0.0, 0.0)
> [ INFO    ] pibooth           : Connected to printer 'Canon_MB2100_series'
> [ DEBUG   ] pibooth           : Activate state 'wait'

Settings

>    OS [Raspbian (buster)]
>    Operating System: Raspbian GNU/Linux 10 (buster)
>    Kernel: Linux 5.10.103-v7+
>    Architecture: arm
>    Camera [Raspberry PI Camera Module 2 + Nikon D750]
>    Pibooth version [2.0.8]

Help would be very much appreciated! I am new to PIBOOTH, but it is great!!! With a workaround/fix for this bug, we would start building the hardware.

Armand-B, 2025-02-03:
Please @werdeil , do you have any idea why hybrid mode isn't triggered ?

werdeil, 2025-02-03:
Hi @Armand-B, I'm not sure about your case but at least @SarkinatorLoramenti's case is "expected"

According to https://github.com/pibooth/pibooth/blob/master/pibooth/camera/__init__.py we don't handle the case Hybrid case Picamera + OpenCV.

Hybrid is planned for OpenCV + GPhoto or PiCamera + GPhoto. We considered that OpenCV camera is a low quality one so we didn't add the PiCamera + OpenCV.

Is it your case? Your DSLR doesn't work with Gphoto?

Armand-B, 2025-02-03:
Hello @werdeil , and thank you very much for your quick answer.
Ok, now I understand how hybrid mode is supposed to work. 
And after a quick check , pibooth is using opencv to use my dslr (Canon EOS 400D). But I can't understand why. 
I tried using it with gphoto2 (gphoto --capture-image-and-download) and it works like a charm. 
So, if my camera is gphoto2 compatible, why is pibooth using it with opencv ?

werdeil, 2025-02-03:
How did you install pibooth ? did you add the `dslr` option when launching pip? if not you can install the gphoto python library afterwards : `pip install gphoto2>=2.0.0` and retry

Armand-B, 2025-02-03:
IIRC I did put the dslr option when I installed pibooth. But maybe it did not work.
I tried installing it again with your command but got this error : 
`Command "/usr/bin/python -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-7YEaWw --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- "setuptools >= 59.6" toml" failed with error code 1 in None`
I'm sorry i'm quite new to python and stuff. Does it mean it is already installed or do I have another issue ?

Armand-B, 2025-02-03:
Ok, as you told me @werdeil , it seems to be a problem with my gphoto2. But while piboith-diag doesn't seem to find gphoto2 installed, when I ask my raspberry, everything seems fin, gphoto2 is well installed. Am I missing something ? 

`pi@raspberrypi:~ $ pibooth-diag
/usr/lib/python3.7/importlib/_bootstrap.py:219: RuntimeWarning: Your system is neon capable but pygame was not built with support for it. The performance of some of your blits could be adversely affected. Consider enabling compile time detection with environment variables like PYGAME_DETECT_AVX2=1 if you are compiling without cross compilation.
  return f(*args, **kwds)
pygame 2.6.1 (SDL 2.0.9, Python 3.7.3)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
Pibooth version installed: 2.0.8
Installed plugins: qrcode-1.0.2, google-photo-1.2.3
gPhoto2 not installed, cannot diagnose connected DSLR
pi@raspberrypi:~ $ gphoto2 --version
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking`

Thanks for your help, and for your work on pibooth, it is really appreciated :)

werdeil, 2025-02-04:
From what I see gphoto is installed but not the python wrapper. That's why gphoto direct commands work but not the one through pibooth.

Can you retry with this command?
```sudo pip3 install gphoto```

Armand-B, 2025-02-04:
I don't know if you intended to write gphoto or you meant gphoto2, but I tried both and got that : 

`pi@raspberrypi:~ $ sudo pip3 install gphoto
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting gphoto
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/gphoto/

pi@raspberrypi:~ $ sudo pip3 install gphoto2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: gphoto2 in /usr/local/lib/python3.7/dist-packages (2.5.1)`

And still, pibooth-diag tell me that I don't have gphoto2...

Armand-B, 2025-02-06:
@werdeil , should I use some kind of interface between python and gphoto2 (like this one https://github.com/jim-easterbrook/python-gphoto2) ? 
I'm not sure if this is what you mean with "python wrapper"

Armand-B, 2025-02-07:
Hello @werdeil , thanks for the answer.
Ok, I did not know that ```pip3 install gphoto2``` was supposed to install the wrapper.

I did try to execute those 2 commands , unfortunately nothing has changed : 
```pi@raspberrypi:~ $ sudo apt install libexif12 libgphoto2-6 libgphoto2-port12 libltdl7
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
libexif12 est déjà la version la plus récente (0.6.21-5.1+deb10u5).
libexif12 passé en « installé manuellement ».
libgphoto2-6 est déjà la version la plus récente (2.5.22-3).
libgphoto2-6 passé en « installé manuellement ».
libgphoto2-port12 est déjà la version la plus récente (2.5.22-3).
libgphoto2-port12 passé en « installé manuellement ».
libltdl7 est déjà la version la plus récente (2.4.6-9).
libltdl7 passé en « installé manuellement ».
Le paquet suivant a été installé automatiquement et n'est plus nécessaire :
  libdav1d3
Veuillez utiliser « sudo apt autoremove » pour le supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
pi@raspberrypi:~ $ pip3 install gphoto2 --user --only-binary :all:
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: gphoto2 in /usr/local/lib/python3.7/dist-packages (2.5.1)```

Even after a reboot pibooth-diag still tell me gphoto2 is not installed.

Armand-B, 2025-02-11:
Any last idea @werdeil ? 
Or maybe @anxuae ? 
I'm stuck on this one.
Anyway, again, thank you for your work and your help :)