pibooth/pibooth #632: add a choice; gif or picture generation !

sebaule, 2025-03-22:
**Is your feature request related to a problem? Please describe.**
There are no gif proposition, on lu pictures

**Describe the solution you'd like**
I think we could had a screen after "take a photo" to ask if we want a picture or a gif, the next screen depends of this choice, if picture like now, Propose 1, or more pinctures and if we select GIF, we propose 2 / 4 or 6 (for exemple) pictures and at the end, we generate a gif ! (no print here) the gif is display !
the gif can have effects or no ! 

**Describe alternatives you've considered**
we could not add a screen and propose pictures or gif on one screen but it is difficult to choice without fingerscreen...
so the best is to add the screen before 👍 

I saw this feature in a wedding, it was really very fun !!!

What do you think of that ?

thanks ! 
Regards 
Seb

werdeil, 2025-03-24:
Hi @sebaule, there is already an option to show an animation instead of the final 4 images layout. Have you tried it?

sebaule, 2025-03-24:
oh ? no ! I will try iy asap ! 
I am waiting a pi 4 with 2g ... 1 g  for Ram is really too short...
I hope in one week ! 
Thx for the tips ! :)

sebaule, 2025-04-02:
@werdeil , if I make a pull request on it, do you think my code can be integrate ? 
of it is not necessary ?

werdeil, 2025-04-02:
I don't think adding a new step is a good idea, but what could be done is if the user choose the animation mode, pibooth save the gif file along the "classic" image and I guess it shall be possible to make it accessible to the gphoto plugin (same name but .gif at the end.

sebaule, 2025-04-03:
I thouhg someting like that : 

[image]
and for exemple something like that too at the first screen : 

[image]

and if you add google photo plugin, you can upload the gif and if you add the qrcode plugin, you can also dowload it ! 
What do you think about it @werdeil  ?