pibooth/pibooth #635: Impression automatique

mememps, 2025-03-30:
Bonjour 
Malgré les réglages dans  les config pibooth lance une impression automatique...
Est-ce lié à utilisation d'une imprimante dnp?

skylu11, 2026-01-07:
Bonjour, 

Je possède également une imprimante DNP mais impossible pour moi de lancer une impression depuis pibooth (ni ma raspberry pi).
Si jamais vous pouviez m'aider sur ce sujet, cela m'avancerait beaucoup.

Merci d'avance.

---------------------------------------
Hello, 

I also own a DNP printer but it’s impossible for me to start printing from Pibooth (nor my raspberry pi).
If you could ever help me on this topic, it would be very beneficial for me.

Thank you in advance.

dominiquegrota, 2026-01-07:
To use a DNP Printer, you need to install cups for printing process. 
Once cups installed, y can use gutenberg drivers : https://gimp-print.sourceforge.io/
When your DNP is default printer, pibooth will print without parameters in pibooth
If you want to print automatically on picture, you can modify le config file and replace 
auto_print = 0
with : 
auto_print = 1