pibooth/pibooth #638: Links on install guide don't seem to point to buster

satcomjimmy, 2025-05-09:
I went through imaging and reinstalling a few times before checking the raspberry pi version. I was getting Bullseye from the link on the docs, and didn't notice. Here is the link to the OS version that ended up working for me. https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-05-28/

baloe, 2025-06-13:
Well, .. buster is from 2021. Won't pibooth run with bullseye?

werdeil, 2026-09-11:
This is fixed in [2.1.0](https://github.com/pibooth/pibooth/releases/tag/2.1.0), though not in the way you asked — the mismatch you ran into no longer exists because the guide no longer targets Buster at all.

The installation page now targets Raspberry Pi OS **Bookworm** (64 bit) with Python 3.10 or higher, and the stale image link, which no longer pointed where the text claimed, is replaced by the official download page. So imaging the system the guide describes gives the system the guide was written for.

Sorry you lost several imaging rounds to that.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-09-11 by werdeil